Seeed-Studio/meta-st-odyssey #3: Kernel didn't work

huuthinh1603, 2022-02-15:
Hi, after following the instructions in the README, then I have built and burned firmware success, but the kernel didn't work. Following is log boot

NOTICE:  CPU: STM32MP157CAC Rev.Z
NOTICE:  Model: STMicroelectronics STM32MP157C-DK2 Discovery Board
INFO:    Reset reason (0x214):
INFO:      IWDG2 Reset (rst_iwdg2)
INFO:    PMIC version = 0x20
INFO:    FCONF: Reading TB_FW firmware configuration file from: 0x2ffe3000
INFO:    FCONF: Reading firmware configuration information for: stm32mp_io
INFO:    Using SDMMC
INFO:      Instance 1
INFO:    Boot used partition fsbl1
NOTICE:  BL2: v2.4-r1.0(debug):v2.4-dirty
NOTICE:  BL2: Built : 16:43:51, Nov 17 2020
INFO:    BL2: Doing platform setup
INFO:    RAM: DDR3-DDR3L 16bits 533000Khz
WARNING: Couldn't find property st,phy-cal in dtb                               
INFO:    Memory size = 0x20000000 (512 MB)                                      
INFO:    BL2: Loading image id 31                                               
INFO:    Loading image id=31 at address 0x2ffff000                              
INFO:    Image id=31 loaded: 0x2ffff000 - 0x2ffff226                            
INFO:    FCONF: Reading FW_CONFIG firmware configuration file from: 0x2ffff000  
INFO:    FCONF: Reading firmware configuration information for: dyn_cfg         
INFO:    FCONF: Reading firmware configuration information for: stm32mp1_firewal
INFO:    BL2: Loading image id 4                                                
INFO:    Loading image id=4 at address 0x2ffc5000                               
INFO:    Image id=4 loaded: 0x2ffc5000 - 0x2ffd847c                             
WARNING: Not a known TEE, use default loading options.                          
INFO:    BL2: Skip loading image id 21                                          
INFO:    BL2: Skip loading image id 22                                          
INFO:    BL2: Loading image id 23                                               
INFO:    Loading image id=23 at address 0xc0500000                              
INFO:    Image id=23 loaded: 0xc0500000 - 0xc0515b2b                            
INFO:    BL2: Loading image id 26                                               
INFO:    Loading image id=26 at address 0x2ffc0000                              
INFO:    Image id=26 loaded: 0x2ffc0000 - 0x2ffc47dc                            
INFO:    BL2: Loading image id 5                                                
INFO:    Loading image id=5 at address 0xc0100000                               
INFO:    Image id=5 loaded: 0xc0100000 - 0xc01d3c48                             
NOTICE:  BL2: Booting BL32                                                      
INFO:    Entry point address = 0x2ffc5000                                       
INFO:    SPSR = 0x1d3                                                           
NOTICE:  SP_MIN: v2.4-r1.0(debug):v2.4-dirty                                    
NOTICE:  SP_MIN: Built : 16:43:51, Nov 17 2020                                  
INFO:    ARM GICv2 driver initialized                                           
INFO:    Set calibration timer to 60 sec                                        
INFO:    stm32mp1 IWDG1 is secure                                               
INFO:    ETZPC: I2C4 (5) could be non secure                                    
INFO:    SP_MIN: Initializing runtime services                                  
INFO:    SP_MIN: Preparing exit to normal world                                 
                                                                                
                                                                                
U-Boot 2020.10-stm32mp-r1 (Oct 05 2020 - 15:15:32 +0000)                        
                                                                                
CPU: STM32MP157CAC Rev.Z                                                        
Model: Seeed Studio Odyssey-STM32MP157C Board                                   
Board: stm32mp1 in trusted mode (seeed,odyssey-stm32mp157c)                     
DRAM:  512 MiB                                                                  
Clocks:                                                                         
- MPU : 650 MHz                                                                 
- MCU : 208.878 MHz                                                             
- AXI : 266.500 MHz                                                             
- PER : 24 MHz                                                                  
- DDR : 533 MHz                                                                 
WDT:   Started with servicing (32s timeout)                                     
NAND:  0 MiB                                                                    
MMC:   STM32 SD/MMC: 0, STM32 SD/MMC: 1                                         
Loading Environment from MMC... *** Warning - bad CRC, using default environment
                                                                                
In:    serial                                                                   
[... 149 lines cut ...]
81 bytes read in 34 ms (2 KiB/s)                                                
Loaded environment from /uEnv.txt                                               
debug: [dtb=stm32mp157c-odyssey.dtb] ...                                        
Using: dtb=stm32mp157c-odyssey.dtb ...                                          
Checking if uname_r is set in /uEnv.txt...                                      
Running uname_boot ...                                                          
npi-uname-booting ....bootpart=0:4                                              
can find /zImage                                                                
loading /zImage ...                                                             
7462795 bytes read in 531 ms (13.4 MiB/s)                                       
debug: [enable_uboot_overlays=] ...                                             
loading /dtbs/5.10.10-stm32-r1/stm32mp157c-odyssey.dtb ...                      
62532 bytes read in 39 ms (1.5 MiB/s)                                           
uboot_overlays: add [enable_uboot_overlays=1] to /boot/uEnv.txt to enable...    
debug: [console=ttySTM0,115200 root=/dev/mmcblk0p5 ro rootfstype=ext4 rootwait].
debug: [bootz 0xC2000000 - 0xC4000000] ...                                      
Kernel image @ 0xc2000000 [ 0x000000 - 0x71dff0 ]                               
## Flattened Device Tree blob at c4000000                                       
   Booting using the fdt blob at 0xc4000000                                     
   Loading Device Tree to cffed000, end cffff443 ... OK                         
                                                                                
Starting kernel ...

Please advise.
Thanks

closed 2022-02-15 by huuthinh1603

lakshanthad, 2022-02-15:
Hello @huuthinh1603,

Could you please share your solution if possible? It will help others if anyone else encounters the same problem as you.

Thank you.

rhaberkorn, 2022-02-21:
@lakshanthad The problem you had is that the kernel is quiet and just hangs, right?
I have the same problem and would also be very interested in the fix. Unfortunately, @huuthinh1603 hasn't pushed anything.

rhaberkorn, 2022-02-21:
Well the other way around, @huuthinh1603 created the issue... You got what I mean. :-D

rhaberkorn, 2022-02-23:
I've also solved the problem. Don't know if it's related with the cause of @huuthinh1603 but I wasn't actually using the meta-st-odyssey layer as it does not support Yocto hardknott and effectively appears to depend on meta-st-openstlinux. I decided it's not worth forking for the time being and imported all the relevant patches from meta-st-odyssey into my own layer. Turns out I was missing the u-boot patches.