Seeed-Studio/meta-st-odyssey #4: No display output with Raspberry Pi 7" Display. Status?

telkamp, 2022-02-17:
I'm using a [Seeed Odyssey NPi](https://wiki.seeedstudio.com/ODYSSEY-STM32MP157C/) with [Raspberry Pi Touch Display](https://www.raspberrypi.com/products/raspberry-pi-touch-display/).
I was able to build Yocto Linux as described in your [README.md](https://github.com/Seeed-Studio/meta-st-odyssey/blob/master/README.md). What is missing is how the vendorfs (with the libraries libEGL.so etc) should be written to the SD card, but this can be copied by hand.

The system boots from SD-card, and the backlight is turned on. But the display shows only a bright surface and no graphics output. But I can scan the ATTiny at i2c bus on the Raspi display board (I see the i2c addresses 0x38 and 0x45), 

When I have look into the patches, I notice that (at least) the touchscreen driver was disabled in device tree, see [0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L41](https://github.com/Seeed-Studio/meta-st-odyssey/blob/324a68e1e3129357d89d6d837a08b8ff498304d1/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L41)
When I enable the driver, I see a Null Pointer:
```
[...]
[   10.068796] systemd[1]: Starting Create Static Device Nodes in /dev...
[   10.082281] rpi_touchscreen_enable be called
[   10.085122] 8<--- cut here ---
[   10.088151] Unable to handle kernel NULL pointer dereference at virtual address 00000000
         Starting Create Static Device Nodes in /dev...
[   10.122218] pgd = 204bec49
[   10.123488] [00000000] *pgd=c32f1835, *pte=00000000, *ppte=00000000
[   10.129775] Internal error: Oops: 17 [#1] PREEMPT SMP ARM
[   10.135141] Modules linked in: sch_fq_codel ipv6
[   10.139741] CPU: 1 PID: 626 Comm: psplash-drm Not tainted 5.10.10 #1
[   10.146079] Hardware name: STM32 (Device Tree Support)
[   10.151228] PC is at i2c_smbus_write_byte_data+0x18/0x70
[   10.156519] LR is at 0x2
[   10.159037] pc : [<c092984c>]    lr : [<00000002>]    psr: 600f0113
[...]
```
What is the current status of the project? Should I see something on the display? What am I doing wrong?

leadercxn, 2022-02-21:
> I'm using a [Seeed Odyssey NPi](https://wiki.seeedstudio.com/ODYSSEY-STM32MP157C/) with [Raspberry Pi Touch Display](https://www.raspberrypi.com/products/raspberry-pi-touch-display/). I was able to [...]

hello, telkamp

Do you have the transfer board which be used to connect Raspberry 7'' Display and odyssey-stm32mp157 ?
like this 
[image]

leadercxn, 2022-02-21:
> 

hello, telkamp

Can you provide a completed log about  it ？

leadercxn, 2022-02-21:
> I'm using a [Seeed Odyssey NPi](https://wiki.seeedstudio.com/ODYSSEY-STM32MP157C/) with [Raspberry Pi Touch Display](https://www.raspberrypi.com/products/raspberry-pi-touch-display/). I was able to [...]

hello, @telkamp 

Can you provide a completed log about it ？

telkamp, 2022-02-22:
Thank you for your response! So it looks that it should work 😄 

I don't have a transfer board. Where can I get it?

Maybe I'm completely wrong, but because the Pins 1-15 are the same on both connectors, I thought that it should be possible to insert the FPC cable Pin1-aligned to NPi's connector J11, as shown here:
[image][image]
So maybe some of the Pins `DSI_LCD_INT`, `DSI_TE`, `DSI_LCD_BL_CTRL` or `DSI_RESET` have to be connected at the NPi side, otherwise the display controller of the STM32MP157C will not work? Do you have a schematic of the transfer board?

If I have a hardware problem, I should not be surprised that the display will show only a bright surface...
Anyway, I'm sending the requested par of the log here (this will come when I enable the touchscreen in the device tree, see [0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L41](https://github.com/Seeed-Studio/meta-st-odyssey/blob/324a68e1e3129357d89d6d837a08b8ff498304d1/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L41)):
```
[...]
[   11.246212] [rpi_touchscreen_get_modes] be call
[   11.362341] rpi_touchscreen_enable be called
[   11.365195] 8<--- cut here ---
[   11.368239] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   11.403104] systemd-journald[578]: Received client request to flush runtime journal.
[   11.428762] pgd = 66ec4c52
[   11.430016] [00000000] *pgd=c342e835, *pte=00000000, *ppte=00000000
[   11.452643] Internal error: Oops: 17 [#1] PREEMPT SMP ARM
[   11.459230] Modules linked in: sch_fq_codel ipv6
[   11.463843] CPU: 1 PID: 639 Comm: psplash-drm Not tainted 5.10.10 #1
[   11.470206] Hardware name: STM32 (Device Tree Support)
[   11.475360] PC is at i2c_smbus_write_byte_data+0x18/0x70
[   11.480631] LR is at 0x2
[   11.483148] pc : [<c092984c>]    lr : [<00000002>]    psr: 600f0113
[   11.489410] sp : c3217cc0  ip : 00000001  fp : 00000001
[   11.494629] r10: c0ca46ac  r9 : c2824a40  r8 : c0e4e9c8
[   11.499875] r7 : c1ffa700  r6 : c1ffa700  r5 : c281b740  r4 : c281b740
[   11.506374] r3 : c3217cd2  r2 : 00000001  r1 : 00000085  r0 : 00000000
[   11.512901] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   11.520029] Control: 10c5387d  Table: c332406a  DAC: 00000051
[   11.525776] Process psplash-drm (pid: 639, stack limit = 0x523cd214)
[   11.532118] Stack: (0xc3217cc0 to 0xc3218000)
[   11.536475] 7cc0: 00000085 c0e4e9c8 c2824a40 c0ca46ac 00000001 c017b1fc c3217cfc c1ffa700
[   11.544651] 7ce0: c1ffa700 c0b7f78c c3217cfc c1104d08 c1dcbc00 c0b845f4 c281b740 c0710080
[   11.552827] 7d00: c281b740 c0b84604 a529f823 00000002 00000000 c1104d08 c282c040 c281b740
[   11.561004] 7d20: c35a8200 c1ffa700 c1ffa700 c0e4e9c8 c2824a40 c0ca46ac 00000001 c0707d60
[   11.569181] 7d40: c1e38840 c35a8200 c1ffa700 c06ea67c 00000000 c35a8200 00000000 c06ce034
[   11.577357] 7d60: c35a8200 c1dcbc00 00000002 00000000 00000000 c32b7900 c34c4380 c06cf1e4
[   11.585534] 7d80: c35a8200 a0c1d3de 00000002 c06cf880 c35a8200 00000000 c1dcbc00 00000000
[   11.593712] 7da0: c30fc9c0 c06d058c c35a8200 00000000 c3217e1c c35a81c0 c30fc9c0 c06cf730
[   11.601890] 7dc0: 00000000 c3217edc c1dcbc00 c06dc8c0 c30fc9c0 c1202460 00000031 c3217edc
[   11.610068] 7de0: c2825040 c282c040 c282c074 00000020 c3216000 c35a81c0 c34c4380 00000000
[   11.618243] 7e00: c30fc9c0 c282c040 c34c4380 00000000 00000000 c35a81c0 00000001 c32c9b00
[   11.626420] 7e20: 00000002 00000005 00000000 00000000 c1e38860 c1dcbd18 00000100 c1dcbc00
[   11.634597] 7e40: cccccccc c1dcbd24 00000023 00000000 c282c074 c06e7978 c3217edc c1104d08
[   11.642772] 7e60: c32b7900 00000000 c1dcbc00 00000002 c06dc6dc c3217edc c32b7900 be8ebbf8
[   11.650949] 7e80: 00000068 c06d7ae0 00000000 c1104d08 c06864a2 00000068 c0c9a104 c06864a2
[   11.659126] 7ea0: c3217edc 000000a2 c32b7900 c06d7d14 0000e200 00000001 c0e50e90 c010fc40
[   11.667303] 7ec0: 00000001 c3217edc 00000068 c335ea80 c06dc6dc 00000051 00000000 0003b40c
[   11.675479] 7ee0: 00000000 00000001 00000023 00000026 00000000 00000000 00000000 00000001
[   11.683656] 7f00: 000061a8 03ba0320 03e603bc 01e00000 01e301e1 000001f4 00000032 00000000
[   11.691833] 7f20: 00000048 78303038 00303834 00000000 00000000 00000000 00000000 00000000
[   11.700009] 7f40: 00000000 00000800 00000255 c0114a58 00000000 c3217f54 c3217f54 c1104d08
[   11.708186] 7f60: c1894900 fffffdfd c335ea80 c06864a2 be8ebbf8 c335ea80 fffffdfd 00000036
[   11.716362] 7f80: 00000000 c0302f38 00000004 be8ebbf8 c06864a2 00000036 c0100264 c3216000
[   11.724538] 7fa0: 00000036 c0100060 00000004 be8ebbf8 00000004 c06864a2 be8ebbf8 00000001
[   11.732716] 7fc0: 00000004 be8ebbf8 c06864a2 00000036 00000026 00000023 00028efa 00000000
[   11.740893] 7fe0: 417da094 be8ebbdc 417c45ef 410c6118 000e0030 00000004 00000000 00000000
[   11.749098] [<c092984c>] (i2c_smbus_write_byte_data) from [<c0710080>] (rpi_touchscreen_i2c_write+0x10/0x34)
[   11.758912] [<c0710080>] (rpi_touchscreen_i2c_write) from [<c0b84604>] (rpi_touchscreen_enable+0x3c/0x2d0)
[   11.768562] [<c0b84604>] (rpi_touchscreen_enable) from [<c0707d60>] (drm_panel_enable+0x28/0xd4)
[   11.777345] [<c0707d60>] (drm_panel_enable) from [<c06ea67c>] (drm_atomic_bridge_chain_enable+0x78/0x98)
[   11.786834] [<c06ea67c>] (drm_atomic_bridge_chain_enable) from [<c06ce034>] (drm_atomic_helper_commit_modeset_enables+0x134/0x268)
[... cut ...]
[   11.870521] [<c0302f38>] (sys_ioctl) from [<c0100060>] (ret_fast_syscall+0x0/0x54)
[   11.878079] Exception stack(0xc3217fa8 to 0xc3217ff0)
[   11.883131] 7fa0:                   00000004 be8ebbf8 00000004 c06864a2 be8ebbf8 00000001
[   11.891311] 7fc0: 00000004 be8ebbf8 c06864a2 00000036 00000026 00000023 00028efa 00000000
[   11.899485] 7fe0: 417da094 be8ebbdc 417c45ef 410c6118
[   11.904538] Code: e28d3012 e1a0c002 e58d1000 e3a0e002 (e1d020b0)
[   13.325656] ---[ end trace 40c69455db0796cf ]---
[...]
```

leadercxn, 2022-02-23:
> Thank you for your response! So it looks that it should work 😄
>
> I don't have a transfer board. Where can I get it?
>
> Maybe I'm completely wrong, but because the Pins 1-15 are the same on both [...]

Hi,telkamp

It looks like the hardware interfaces are compatible. I have two questions
1. Can you see "rpi_touchscreen_dsi_probe be called" this log in your STM32MP157 boot log?
2. Have you add this node in your dts [https://github.com/Seeed-Studio/meta-st-odyssey/blob/324a68e1e3129357d89d6d837a08b8ff498304d1/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L99](https://github.com/Seeed-Studio/meta-st-odyssey/blob/324a68e1e3129357d89d6d837a08b8ff498304d1/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L99)

It looks like  `mp_ts = devm_kzalloc(dev, sizeof(*mp_ts), GFP_KERNEL);`  this code have not been called, 
[image]
please check this dsi-node in your dts.

Thanks~

telkamp, 2022-02-23:
> Can you see "rpi_touchscreen_dsi_probe be called" this log in your STM32MP157 boot log?

Yes, this is called!
> Have you add this node in your dts...

Yes, I have!
> It looks like mp_ts = devm_kzalloc(dev, sizeof(*mp_ts), GFP_KERNEL); this code have not been called,

**Please note!!!** In contrast to your code, I included the node
[`touchscreen@45`](https://github.com/Seeed-Studio/meta-st-odyssey/blob/324a68e1e3129357d89d6d837a08b8ff498304d1/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0001-Modify-files-to-make-raspberrypi-touchscreen-run-on-.patch#L41) for test. This caused the null pointer I've attached in my previous message. But I'm sure now that the touch input device with the node
```
&i2c1 {
/*
    touchscreen@45 {
        compatible = "edt,edt-ft5406";
*/
[...]
```
is also not working for you - this seems to be another problem.
We should first solve the problem with the bright display. Therefore, I've reverted my changes and I'm using  **exactly**  the code from you master Branch. It still see the bright display and I'm sending you the full log output of the kernel when booting: 
[boot.log](https://github.com/Seeed-Studio/meta-st-odyssey/files/8127629/boot.log)

Do you have an idea? Maybe you have a schemativ of your transfer board so I can check if one of the additional pins should be connected on the Seeed NPi side?

telkamp, 2022-02-23:
I found the problem: My display is broken, so it shows this bright screen also on a Raspberry Pi... Fortunately I have another display, and this shows the same screen than your does in your photo.
Thank you for your help!!!
Is it correct that the capacitive touch is not supported currently?

leadercxn, 2022-02-24:
> I found the problem: My display is broken, so it shows this bright screen also on a Raspberry Pi... Fortunately I have another display, and this shows the same screen than your does in your photo. Thank you for your help!!! Is it correct that the capacitive touch is not supported currently?

Hi, @telkamp 

I'm great to hear you solved the problem.
yes, our raspberrypi driver code does not support the capacitive touch function yet.

thanks ~

telkamp, 2022-02-24:
Many thanks for your support! Even if the problem was on my side, you helped me much to identify the cause.