Seeed-Studio/meta-st-odyssey #5: Layer doesn't support "STM32_PM_SHUTDOWN" mode

bSquaredOlea, 2022-03-01:
If I change "pwr_regulators" in tf-a-stmp32/fdts/stm32mp157c-odyssey.dtsi to support "SHUTDOWN" (set system_off_soc_mode = <STM32_PM_SHUTDOWN>), I get a WDT timeout when executing "shutdown -h 0" in linux.  This prevents one from getting into VBAT mode for a really low power system.
[stm32mp157c-odyssey.dtsi.txt](https://github.com/Seeed-Studio/meta-st-odyssey/files/8163277/stm32mp157c-odyssey.dtsi.txt)

leadercxn, 2022-03-08:
Hello bSquaredOlea ,

I get the same log with system_off_soc_mode = <STM32_PM_CSTOP_ALLOW_STANDBY_DDR_OFF> and system_off_soc_mode = < STM32_PM_SHUTDOWN > , when I executing "shutdown -h 0".
[compara_system_off_soc_mod_log.txt](https://github.com/Seeed-Studio/meta-st-odyssey/files/8205734/compara_system_off_soc_mod_log.txt)

leadercxn, 2022-03-08:
> If I change "pwr_regulators" in tf-a-stmp32/fdts/stm32mp157c-odyssey.dtsi to support "SHUTDOWN" (set system_off_soc_mode = <STM32_PM_SHUTDOWN>), I get a WDT timeout when executing "shutdown -h 0" in [...]

Hello @bSquaredOlea 

I get the same log with system_off_soc_mode = <STM32_PM_CSTOP_ALLOW_STANDBY_DDR_OFF> and system_off_soc_mode = < STM32_PM_SHUTDOWN > , when I executing "shutdown -h 0".
[compare_system_off_soc_mod_log.txt](https://github.com/Seeed-Studio/meta-st-odyssey/files/8205778/compare_system_off_soc_mod_log.txt)

bSquaredOlea, 2022-03-13:
In the latter case, I get a WD timeout and the system reboots.  I  "fixed"
this by patching with this:

index 962a99224..ec99257bf 100644
--- a/plat/st/stm32mp1/stm32mp1_pm.c
+++ b/plat/st/stm32mp1/stm32mp1_pm.c
@@ -205,6 +205,8 @@ static void __dead2 stm32_system_off(void)
                gicv2_raise_sgi(ARM_IRQ_SEC_SGI_1, STM32MP_SECONDARY_CPU);
        }

+       INFO("Olea: sytem Off: ");
+
        stm32_enter_low_power(soc_mode, 0);

        stm32_pwr_down_wfi(false);

Just a print statement.  I went to look at latest ST code, and there are
significant differences.    Any plans to backport TF-A code to Odyssey?

Thanks,
Ben

leadercxn, 2022-03-28:
> 

@bSquaredOlea hi, We have no anyplans to backport TF-A code to Odyssey-STM32MP157.   And I can't recurrence the phenomenon you said. Can you provide some log information?

Thanks～

leadercxn, 2022-03-28:
> In the latter case, I get a WD timeout and the system reboots. I "fixed" this by patching with this: index 962a99224..ec99257bf 100644 --- a/plat/st/stm32mp1/stm32mp1_pm.c +++ [...]

@bSquaredOlea hi, We have no anyplans to backport TF-A code to Odyssey-STM32MP157. And I can't recurrence the phenomenon you said. Can you provide some log information?

Thanks～

closed 2022-05-23 by leadercxn