Seeed-Studio/meta-st-odyssey #7: One of the USB ports is dead

rhaberkorn, 2022-08-03:
The "lower" of the USB ports (the one closer to the PCB) is dead in images produced according to your README. While it receives power, devices plugged into this port are not detected. The other "upper" USB port works fine.
I can also reproduce this in other projects based on the Odyssey SOM and the patches included in this repository.

On the [Debian-based image provided by Seeed](https://files.seeedstudio.com/linux/ODYSSEY%E2%80%93STM32MP157C/stm32mp1-debian-buster-console-armhf-latest-2gb.img.xz), both USB ports work properly. They use an older 4.x kernel and probably other patches. I was doing all tests on the same Odyssey board, so I can exclude hardware issues.

rhaberkorn, 2022-09-16:
[Enabling full-speed USB OTG](https://wiki.st.com/stm32mpu/wiki/OTG_device_tree_configuration#DT_configuration_using_full-speed_USB_PHY) (should enable OTG on the board's USB C port), also did not fix this issue. Neither did I get USB OTG to work in addition to the two USB host ports, but that's another issue.

ALTracer, 2022-09-22:
@rhaberkorn Do you understand USBOTG-FS works on a separate FS PHY on GPIOs PA11-PA12? It is not routable through UTMI+ to USBPHYC. You could use USBOTG-HS on one of HS ports, or use both as EHCI hosts.

I faced a similar issue when bringing up a custom design of STM32MP157CAC on daughterboard and custom motherboard; even though I provided the ST's device-tree declarations, the dwc2 driver simply reports a core hang of usbotg, while both USBPHYC HS ports work as EHCI perfectly. Seems like some logic is missing from the driver, did you try the patch from ST forums https://community.st.com/s/question/0D53W00000UA60XSAT/mainline-linux-fails-to-set-up-fs-otg?t=1663868112035 ?

I'm rebuilding the 5.10.10-stm32mp1-r1 kernel to try to get three ports out of this design. Neither 5.10 nor 5.15 branches include this code to switch to FS PHY for the USBOTG IP core.

bSquaredOlea, 2022-09-23:
Following.  I'm working on getting the same resources enabled on my Yocto image.

rhaberkorn, 2022-09-24:
@ALTracer 
> Do you understand USBOTG-FS works on a separate FS PHY on GPIOs PA11-PA12?

Yes. I was told by my colleagues that they are connected with the USB C port on the Odyssey board. Anyway, that's already the second step after sorting out the driver issue.

> did you try the patch from ST forums https://community.st.com/s/question/0D53W00000UA60XSAT/mainline-linux-fails-to-set-up-fs-otg?t=1663868112035 ?

Indeed I tried. Unsuccessfully.

Anyway, USB OTG is off-topic here. I suggest we continue discussion in the [Seeed Studio forum](https://forum.seeedstudio.com/t/odyssey-stm32mp157c-cannot-activate-full-speed-usb-otg/266147/2).

ALTracer, 2022-09-24:
> Anyway, USB OTG is off-topic here.

Oh, I misunderstood you.

I took a look at the schematics for Seeed Odyssey NPi. Indeed, the two USB-A ports are routed by your vendor to USBPHYC for EHCI hosts, apparently, and USB-C port is using PA11/PA12 aka USBOTG dedicated FS PHY.

My question then changes. Did it work on old OS snapshots and this is a regression? Because the device-tree is incomplete in this repository. It routes one port to the EHCI and other to USBOTG, while missing FS PHY altogether. The patch by @leadercxn added in 0b93e8b7688125f3ac96103001721c5bdcf98e79 adds default [wrong] bindings.
https://github.com/Seeed-Studio/meta-st-odyssey/blob/9c7bedd1fc343de3ddda2adc0ee1ef82c509c40c/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0002-fix_usb_ethernet_bug.patch#L193-L201
Should be
```c
&usbh_ehci {
	phys = <&usbphyc_port0>, <&usbphyc_port1 1>;
	phy-names = "usb", "usb";
	status = "okay";
};

&usbh_ohci {
	phys = <&usbphyc_port0>, <&usbphyc_port1 1>;
	phy-names = "usb", "usb";
	status = "okay";
};
```
per https://wiki.st.com/stm32mpu/wiki/USBH_device_tree_configuration#DT_configuration_when_using_the_two_physical_ports

Also https://github.com/Seeed-Studio/meta-st-odyssey/blob/9c7bedd1fc343de3ddda2adc0ee1ef82c509c40c/recipes-kernel/linux/linux-stm32mp/5.10/5.10.10/0002-fix_usb_ethernet_bug.patch#L203-L211
Should be
```c
&usbotg_hs {
	compatible = "st,stm32mp15-fsotg", "snps,dwc2";
	pinctrl-0 = <&usbotg_hs_pins_a &usbotg_fs_dp_dm_pins_a>;
	pinctrl-names = "default";
	vbus-supply = <&vbus_otg>;
	dr_mode = "otg";
	status = "okay";
};
```
per https://wiki.st.com/stm32mpu/wiki/OTG_device_tree_configuration#DT_configuration_example_as_full-speed_OTG-2C_with_micro-B_connector

I wrote my device trees like this, and our custom device happily works with two HS ports in Host mode. Note that not a single mainline STM32MP15x device tree does it like this. Maybe we should nag them about it as well. Why are all the USB bindings missing from https://github.com/torvalds/linux/blob/master/arch/arm/boot/dts/stm32mp157c-odyssey-som.dtsi ?

P.S. I don't use any Seeed Odyssey products and don't have an account on their forums.

rhaberkorn, 2022-09-24:
@ALTracer Thanks, that did the trick to fix the second USB port.
This should probably be fixed upstream by updating 0002-fix_usb_ethernet_bug.patch.
But I leave this to the maintainers and will close this ticket.

USB OTG I still cannot get to work.

closed 2022-09-24 by rhaberkorn

bSquaredOlea, 2022-11-02:
Did anyone get the OTG to work yet?  I have added the patch and the .dts changes, but still not working for me.