Seeed-Studio/meta-st-odyssey #9: Bootloader Logging Issue

lovelyyyyyyy, 2025-02-03:
HI, we have trying to build img to ST board but as shown on seed studio site credentials are not matching while login the boot login. any suggestions!
sharing the link for site where procedure is stated.-https://wiki.seeedstudio.com/ODYSSEY-STM32MP135D/

mingzhangqun, 2025-02-08:
@lovelyyyyyyy have you tried "grove" ? (Configured by [BR2_TARGET_GENERIC_ROOT_PASSWD](https://github.com/xogium/buildroot-stm32mp135d-odyssey/blob/02e3f1e777d145604f31cbc07f63b89e3c872e49/configs/stm32mp135d_odyssey_defconfig#L8))

lovelyyyyyyy, 2025-02-10:
Now password is correct but how to onn screen(RGB666) on STM32MP135daf7?

mingzhangqun, 2025-02-17:
It needs to enable ltdc in [stm32mp135d-odyssey-grove.dts](https://github.com/xogium/buildroot-stm32mp135d-odyssey/blob/master/board/stm32mp135d-odyssey/dts/stm32mp135d-odyssey-grove.dts), and you can refer with this [stm32mp135d-odyssey.dts#L350](https://github.com/Seeed-Studio/st-linux/blob/8622abaafceca2b5b1d02c49ec4bcd39cc0da72a/arch/arm/boot/dts/stm32mp135d-odyssey.dts#L350).

closed 2025-03-14 by mingzhangqun